jbehley/SuMa #3: terminate called after throwing an instance of 'rv::XmlError'

hunkyu, 2018-07-28:
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1060/PCIe/SSE2
terminate called after throwing an instance of 'rv::XmlError'
  what():  Error while parsing in line 1

jbehley, 2018-07-30:
This usually happens when the XML configuration file is malformed. Ensure that the xml files are well formed.

Try a validation tool, like https://www.xmlvalidation.com/, to validate the xml.

If the error is still thrown, even with a valid xml file, let me know and I will try to figure out what's the reason.

hunkyu, 2018-08-01:
@jbehley  I checked the XML file without a problem
I execute./visualizer directly in the bin directory

jbehley, 2018-08-01:
Hmmm, it's hard to guess what might be the problem.

Either you try to directly call the `visualizer` with the path to the configuration file, i.e., 

`./visualizer /path/to/config/default.xml`

if this does not help, could you start the `visualizer` with gdb and report the backtrace, i.e., 

1. `gdb ./visualizer`
This opens the debugger, then
2. `(gdb) run`
To start the program, then
3. `(gdb) bt`
To get the backtrace, which gives us an indication where the problem originated.

It seems that there is some problem with parsing the `../config/default.xml`. But I unfortunately have no clue where the problem lies.

hunkyu, 2018-08-02:
@jbehley According to what you say, I can use it already.
Can this map be saved to PCD or ply file?

jbehley, 2018-08-02:
This is currently not supported.

How did you solve the problem with the xml? Did you start the visualizer explicitly with the path to the  configuration xml?

hunkyu, 2018-08-02:
yes !  I use directly ./visualizer /path/to/config/default.xml  

Is there any way I can export 3D map model?

jbehley, 2018-08-02:
As mentioned earlier.

We do not support exporting of the map.

closed 2018-09-21 by jbehley