jbehley/SuMa #4: Unknown CMake command "COMPILE_SHADERS"

xiesc, 2018-09-19:
The errors occurred when I compile the program in Ubuntu 16.04
xiesc@xiesc-ThinkPad-T460p:~/xieshichao/surfel_based_mapping$ catkin build --save-config -i --cmake-args -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/xiesc/xieshichao/surfel_based_mapping
--------------------------------------------------------------------------------
Source Space:       [exists] /home/xiesc/xieshichao/surfel_based_mapping/src
Log Space:          [exists] /home/xiesc/xieshichao/surfel_based_mapping/logs
Build Space:        [exists] /home/xiesc/xieshichao/surfel_based_mapping/build
Devel Space:        [exists] /home/xiesc/xieshichao/surfel_based_mapping/devel
Install Space:      [unused] /home/xiesc/xieshichao/surfel_based_mapping/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=430 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> surfel_mapping                                                   
[surfel_mapping:loadenv] Loading environment from: /home/xiesc/xieshichao/surfel_based_mapping/devel/env.sh
[surfel_mapping:cmake] Not searching for unused variables given on the command line.
[surfel_mapping:cmake] -- Boost version: 1.58.0                                 
[surfel_mapping:cmake] -- Found the following Boost libraries:                  
[surfel_mapping:cmake] --   filesystem                                          
[surfel_mapping:cmake] --   system                                              
[surfel_mapping:cmake] -- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include
[surfel_mapping:cmake] -- Using CATKIN_DEVEL_PREFIX: /home/xiesc/xieshichao/surfel_based_mapping/devel/.private/surfel_mapping
[surfel_mapping:cmake] -- Using CMAKE_PREFIX_PATH: /home/xiesc/xieshichao/surfel_based_mapping/devel;/opt/ros/kinetic
[surfel_mapping:cmake] -- This workspace overlays: /home/xiesc/xieshichao/surfel_based_mapping/devel;/opt/ros/kinetic
[surfel_mapping:cmake] -- Using PYTHON_EXECUTABLE: /usr/bin/python              
[surfel_mapping:cmake] -- Using Debian Python package layout                    
[surfel_mapping:cmake] -- Using empy: /usr/bin/empy                             
[surfel_mapping:cmake] -- Using CATKIN_ENABLE_TESTING: ON                       
[surfel_mapping:cmake] -- Call enable_testing()                                 
[surfel_mapping:cmake] -- Using CATKIN_TEST_RESULTS_DIR: /home/xiesc/xieshichao/surfel_based_mapping/build/surfel_mapping/test_results
[surfel_mapping:cmake] -- Found gmock sources under '/usr/src/gmock': gmock will be built
[surfel_mapping:cmake] -- Found gtest sources under '/usr/src/gmock': gtests will be built
[surfel_mapping:cmake] -- Using Python nosetests: /usr/bin/nosetests-2.7        
[surfel_mapping:cmake] -- catkin 0.7.14                                         
[surfel_mapping:cmake] -- Could NOT find glow (missing: glow_DIR)               
[surfel_mapping:cmake] Enabling Nvidia OpenGL extensions.                       
[surfel_mapping:cmake] CMake Error at /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa-master/CMakeLists.txt:54 (COMPILE_SHADERS):
[surfel_mapping:cmake]   **Unknown CMake command "COMPILE_SHADERS"**.               
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake] -- Using OpenGL version 450.                             
[surfel_mapping:cmake] -- Configuring incomplete, errors occurred!              
[surfel_mapping:cmake] See also "/home/xiesc/xieshichao/surfel_based_mapping/build/surfel_mapping/CMakeFiles/CMakeOutput.log".
[surfel_mapping:cmake] See also "/home/xiesc/xieshichao/surfel_based_mapping/build/surfel_mapping/CMakeFiles/CMakeError.log".
Failed    <<< surfel_mapping                [ 0.8 seconds ]                    
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 0.9 seconds total.

xiesc, 2018-09-20:
This problem has been solved since the cmake didn't find the GlowShaderCompilation.cmake.
However, there was another error occurred.

In file included from /usr/local/include/gtsam/nonlinear/NonlinearFactorGraph.h:24:0,
[surfel_mapping:make]                  from /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.h:5,
[surfel_mapping:make]                  from /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/SurfelMapping.h:21,
[surfel_mapping:make]                  from /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/SurfelMapping.cpp:1:
[surfel_mapping:make] /usr/local/include/gtsam/geometry/Point2.h: In static member function ‘static gtsam::Vector gtsam::Point2::Logmap(const gtsam::Point2&)’:
[surfel_mapping:make] /usr/local/include/gtsam/geometry/Point2.h:175:86: error: could not convert ‘Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseBase<Derived>::Scalar = double]((& dp)->gtsam::Point2::x()).Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, 1> >((& dp)->gtsam::Point2::y())’ from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘gtsam::Vector {aka Eigen::Matrix<double, -1, 1>}’
[surfel_mapping:make]    static inline Vector Logmap(const Point2& dp) { return (Vector(2) << dp.x(), dp.y()); }                                                                                            
[surfel_mapping:make]                                                                                       ^
[surfel_mapping:make] /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.cpp: In member function ‘std::vector<Eigen::Matrix<double, 4, 4> > Posegraph::poses() const’:
[surfel_mapping:make] /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.cpp:73:34: error: ‘const class gtsam::Value’ has no member named ‘cast’
[surfel_mapping:make]      poses[pair.key] = pair.value.cast<Pose3>().matrix();                                                                                                                             
[surfel_mapping:make]                                   ^
[surfel_mapping:make] /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.cpp:73:44: error: expected primary-expression before ‘>’ token
[surfel_mapping:make]      poses[pair.key] = pair.value.cast<Pose3>().matrix();                                                                                                                             
[surfel_mapping:make]                                             ^
[surfel_mapping:make] /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.cpp:73:46: error: expected primary-expression before ‘)’ token
[surfel_mapping:make]      poses[pair.key] = pair.value.cast<Pose3>().matrix();                                                                                                                             
[surfel_mapping:make]                                               ^
[surfel_mapping:make] [ 70%] Building CXX object CMakeFiles/suma.dir/suma_autogen/mocs_compilation.cpp.o                                                                                                    
[surfel_mapping:make] test/CMakeFiles/test_posegraph.dir/build.make:62: recipe for target 'test/CMakeFiles/test_posegraph.dir/__/src/core/Posegraph.cpp.o' failed                                           
[surfel_mapping:make] make[2]: *** [test/CMakeFiles/test_posegraph.dir/__/src/core/Posegraph.cpp.o] Error 1                                                                                                 
[surfel_mapping:make] CMakeFiles/Makefile2:824: recipe for target 'test/CMakeFiles/test_posegraph.dir/all' failed                                                                                           
[surfel_mapping:make] make[1]: *** [test/CMakeFiles/test_posegraph.dir/all] Error 2                                                                                                                         
[surfel_mapping:make] make[1]: *** Waiting for unfinished jobs....                                                                                                                                          
[surfel_mapping:make] CMakeFiles/suma.dir/build.make:109: recipe for target 'CMakeFiles/suma.dir/src/core/SurfelMapping.cpp.o' failed                                                                       
[surfel_mapping:make] make[2]: *** [CMakeFiles/suma.dir/src/core/SurfelMapping.cpp.o] Error 1                                                                                                               
[surfel_mapping:make] make[2]: *** Waiting for unfinished jobs....                                                                                                                                          
[surfel_mapping:make] /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.cpp: In member function ‘std::vector<Eigen::Matrix<double, 4, 4> > Posegraph::poses() const’:
[surfel_mapping:make] /home/xiesc/xieshichao/surfel_based_mapping/src/SuMa/src/core/Posegraph.cpp:73:34: error: ‘const class gtsam::Value’ has no member named ‘cast’
[surfel_mapping:make]      poses[pair.key] = pair.value.cast<Pose3>().matrix();                                                                                                                             
[... cut ...]
[surfel_mapping:make] CMakeFiles/Makefile2:240: recipe for target 'CMakeFiles/suma.dir/all' failed                                                                                                          
[surfel_mapping:make] make[1]: *** [CMakeFiles/suma.dir/all] Error 2                                                                                                                                        
[surfel_mapping:make] Makefile:140: recipe for target 'all' failed                                                                                                                                          
[surfel_mapping:make] make: *** [all] Error 2

jbehley, 2018-09-20:
Dear @xiesc,

the cmake command `COMPILE_SHADER` is defined in the file "GlowShaderCompilation.cmake", which is part of glow (https://github.com/jbehley/glow). Usually, the cmake file should be available via the catkin environment.

For gtsam 4.0, I had to build gtsam with the option `GTSAM_USE_SYSTEM_EIGEN` set to `ON` to get it working with my code.

hope that solves your problems.

xiesc, 2018-09-20:
Dear @jbehley 

Thank you very much for your reply. The build is done since I re-compile the gtsam 4.
The **visualizer**  works well but sometimes the OS system corrupts when I terminate the visualizer terminal.

I read your paper just now, and I have a question about the TABLE 2 and the Fig. 6. For sequence 08, the Frame-to-Model with loop closure path(green) seems better from Fig.6. But the loss of it is 1.2% which is bigger than the Frame-to-Model without loop closure of 1.0%. So, I wander that is this situation caused by the "z axis" error which cannot be  shown in Fig.6?

By the way, I also test the LOAM on training date set. The result of the translation error is : 0.77 0.96 1.09 0.99 0.82 0.43 0.77 0.59 2.25 1.67 1.79 percentage (the criterion provided by KITTI). 
And I'm very interested in your research since I think the surfel is an idea representation of the map since it is also convenience for updating map by other sensors like monocular camera or RGBD camera. And I'm working on the semantic SLAM based on your research since I think the semantic information provided by camera will solve some backwards mentioned in your article.
Sincerely.

jbehley, 2018-09-21:
Regarding the "corruption": Try to reduce the size of the history buffer. Usually this helps to avoid the problem.

Regarding the comment on the performance: As already mentioned in the paper, the KITTI odometry metric measures only relative errors over 800 m, but does not account for global consistency. Thus, you have to expect that you get better numbers if your reproduce the ground truth.

However, the ground truth is generated from an inertial navigation system, combining GPS + inertial measurements, which is also affected by drift. Therefore, you will also see errors in the height of the trajectory and not always the correctly loop closed trajectory. (Just plot the ground truth poses from sequence 08 with a 3d plot and convince yourself or see the trajectory from sequence 00, where the car enters the same streets.)

xiesc, 2018-09-21:
It true that the INS system is affected by drift, but the LiDAR odometry results seems extremely serious in height. When I use SuMa running on the sequence 01, the height has drifted to -300 meters.
It also happened in other LiDAR based methods like LOAM or NDT. I have solved this problem by some tricks, and the height error for LOAM has became less than 0.8%. And I'm trying to apply it on SuMa, the better accuracy can be achieved I hope. And I also want to know is the height drifting so much in your experiments?
Thank you very much, and sorry for my poor English.  OrT

jbehley, 2018-09-26:
Sorry, I missed your last comment.

For the highway scene (sequence 01), yes, I also experienced the drift in z-direction. I suspect that this comes mainly from the "sparsity" in vertical direction.