jbehley/SuMa #5: Invalid framebuffer object. Code 36061(GL_FRAMEBUFFER_UNSUPPORTED)

kongcheng08, 2018-10-17:
Hello, I run kiitti dataset and suffer from this problem when play second scan:

 reason Invalid framebuffer object. Code 36061(GL_FRAMEBUFFER_UNSUPPORTED)
terminate called after throwing an instance of 'glow::GlFramebufferError'
  what():  Invalid framebuffer object. Code 36061(GL_FRAMEBUFFER_UNSUPPORTED)

and gdb show:
(gdb) bt
#0  0x00007ffff3fb2428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff3fb402a in __GI_abort () at abort.c:89
#2  0x00007ffff45ec84d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff45ea6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff45ea701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff45ea919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff793334a in glow::GlFramebuffer::bind() () from /home/fjy/code/offline/suma_ws/devel/.private/glow/lib/libglow.so
#7  0x00007ffff7b8c015 in Frame2Model::jacobianProducts (this=0x230cfa0, JtJ=..., Jtf=...) at /home/fjy/code/offline/suma_ws/src/SuMa/src/core/Frame2Model.cpp:176
#8  0x00007ffff7ba2de8 in LieGaussNewton::step (this=this@entry=0x2393070) at /home/fjy/code/offline/suma_ws/src/SuMa/src/core/LieGaussNewton.cpp:60
#9  0x00007ffff7ba3d6f in LieGaussNewton::minimize (this=0x2393070, F=..., T0=...) at /home/fjy/code/offline/suma_ws/src/SuMa/src/core/LieGaussNewton.cpp:30
#10 0x00007ffff7b73c1b in SurfelMapping::updatePose (this=this@entry=0x1396890) at /home/fjy/code/offline/suma_ws/src/SuMa/src/core/SurfelMapping.cpp:396
#11 0x00007ffff7b77e27 in SurfelMapping::processScan (this=0x1396890, scan=...) at /home/fjy/code/offline/suma_ws/src/SuMa/src/core/SurfelMapping.cpp:191
#12 0x0000000000478610 in VisualizerWindow::setScan (this=0x7fffffffcbf0, idx=<optimized out>) at /home/fjy/code/offline/suma_ws/src/SuMa/src/visualizer/VisualizerWindow.cpp:728
#13 0x00007ffff4b93d2a in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff57f285e in QAbstractSlider::valueChanged(int) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff55586ed in QAbstractSlider::setValue(int) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00000000004721a6 in VisualizerWindow::nextScan (this=0x7fffffffcbf0) at /home/fjy/code/offline/suma_ws/src/SuMa/src/visualizer/VisualizerWindow.cpp:609
#17 0x00007ffff4b93d2a in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff4ba05c8 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff4b94bb3 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff545805c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff545d516 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff4b6538b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff4bba5ed in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff4bbaaf1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007fffed192197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007fffed1923f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007fffed19249c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff4bbb7cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff4b62b4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff4b6abec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x000000000042d25e in main (argc=1, argv=0x7fffffffd498) at /home/fjy/code/offline/suma_ws/src/SuMa/src/visualizer/visualizer.cpp:41

error occur in    core/Frame2Model.cpp   fbo_blend_.bind();

before  fbo_blend_.bind()

 fbo_blend_.attach(FramebufferAttachment::COLOR0, JtJJtf_blend_); 
return  fbo_blend_.valid() is false
and                                
fbo_blend_.attach(FramebufferAttachment::DEPTH_STENCIL, rbo2);
return  fbo_blend_.valid() is also false

and how can solve

jbehley, 2018-10-19:
Hi @kongcheng08,

do you use an Intel GPU for computing/visualizing? 

It could be that Intel (and AMD?) are more pedantic in implementing the core profile. It might be that the format RGB_FLOAT of the framebuffer is not supported by your graphics card. Therefore, I have to change something at the code  to make it possible to run it on non-NVIDIA GPUs.

DavidJamesOKC, 2018-10-25:
Hello, when I run the Kitti dataset and SuMa, I found some problems. 
When I imported a frame of data, it showed "Error: KIT robot model not found." and then I clicked the"play button" and the terminal was over.
How can I solve? I'm looking forward to your reply. Thanks.

ClaudioCimarelli, 2018-11-27:
@jbehley, same problem here with intel UHD 620 :(

pangfumin, 2019-03-05:
same problem. waiting for fix

jbehley, 2019-03-05:
@pangfumin, @ClaudioCimarelli and @kongcheng08,

Sorry that you had to wait, but there were always some more pressing issues.

I just commited a fix that hopefully fixes your issues, but I cannot test it here, since I have a NVIDIA GPU, which supports all Framebuffer types (even though it is not required by the core profiles specification) Intel seems to be more strict with the core profile specification.

Let me know if this solves the issue. (And hopefully it did not break something else. ;P)

pangfumin, 2019-04-29:
@jbehley Thank you for your support. This commit fixs the GL_FRAMEBUFFER_UNSUPPORTED error on my Intel graphic card. Great Work!

jbehley, 2019-04-29:
Thanks for the info!

Then I consider this issue solved and close it. :)

closed 2019-04-29 by jbehley