jbehley/SuMa #6: Error: KIT robot model not found

DavidJamesOKC, 2018-10-25:
Hello, when I run the Kitti dataset and SuMa, I found some problems.
When I imported a frame of data, it showed "Error: KIT robot model not found." and then I clicked the"play button" and the terminal was over.
How can I solve? I'm looking forward to your reply. Thanks.

ClaudioCimarelli, 2018-11-26:
Dear @jbehley, should we create our own model of the car?
I tried to load a free model quickly, but I had a problem when loading the material.

jbehley, 2018-11-26:
No, @ClaudioCimarelli. The model is only for visualization purposes. The model is inside the asset folder and you can try to  replace the relative path with an absolute path. (The loading is iniside `VisualizerWindow::initilizeKITTIrobot()` , see `ViewportWidget.cpp`)

The obj  loader is some custom made "minimal version" that may not support all specifics of the obj format.

ClaudioCimarelli, 2018-11-27:
@jbehley, thanks for your quick response.

The fact is that I saw inside the class you mentioned and the path references an ".obj" file. Instead, in "assets" folder there are only ".mtl" files. Sorry, I do not understand this.

jbehley, 2018-11-27:
@ClaudioCimarelli and @DavidJamesOKC, I  now figured out what was the problem. You were totally right that the files were missing.

I did not recognize this, since they were locally there, but were not added to the commit as "*.obj" is inside the `.gitignore` file. I added now the missing files and now it should work.

Thanks for pointing this out. :+1:

ClaudioCimarelli, 2018-11-27:
Thanks a lot, i commented out the line of code calling that method meanwhile. 
A little question, why if in the option there is not the parameter "show robot" the thick is set true in any case?

jbehley, 2018-11-27:
The code is somehow "research code" or in a proof of concept state. Thus, some things are not that nicely implemented. (A lot of the code is just there to get some nice pictures for the paper and plots. ;)) And the code does definitely not cover all cases or circumstances.

It would be nicer to directly disable  (and uncheck) the checkbox if no model was found, but this case is currently not covered by my visualization tool. A lot of the failure cases are also not covered and some of the classes are just to big and handle to much stuff. Thus, a major refactoring would be needed to get the code in a good shape. But I still hope that the code is enough to get the main ideas and try out  something. Even though it might be not perfectly engineered.

ClaudioCimarelli, 2018-11-28:
@jbehley, i did not mean to criticize your work. It was more my curiosity on how this function could work, because I am not expert of ui, specially with c++. However, this is a great example for me to reproduce other uis by following this code as a footprint. Thanks again for your time.

P.S.: I think you could close this issue if it's solved, I will avoid to spam further.

jbehley, 2018-11-28:
Thanks for your comments and for pointing me at the bug. And I just wanted to clarify that the code is somehow in a state of "good enough". There are ample opportunities for improvement. ;)

 If you have questions, don't hesitate to ask me.

closed 2018-11-28 by jbehley