jbehley/SuMa #9: "1 package failed" when i do catkin build

Lyusijie, 2019-02-28:
Hi, when i do " catkin build --save-config -i --cmake-args -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=430 -DENABLE_NVIDIA_EXT=YES",there shows 1 package failed
here is the specific error:
[surfel_mapping:cmake] CMake Error at /home/lvsijie/catkin_ws/src/SuMa/CMakeLists.txt:175 (add_executable):
[surfel_mapping:cmake]   Target "genvideo" links to target "Boost::serialization" but the target was
[surfel_mapping:cmake]   not found.  Perhaps a find_package() call is missing for an IMPORTED
[surfel_mapping:cmake]   target, or an ALIAS target is missing?                 
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake] CMake Error at /home/lvsijie/catkin_ws/src/SuMa/CMakeLists.txt:175 (add_executable):
[surfel_mapping:cmake]   Target "genvideo" links to target "Boost::thread" but the target was not
[surfel_mapping:cmake]   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
[surfel_mapping:cmake]   an ALIAS target is missing?                            
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake] CMake Error at /home/lvsijie/catkin_ws/src/SuMa/CMakeLists.txt:175 (add_executable):
[surfel_mapping:cmake]   Target "genvideo" links to target "Boost::date_time" but the target was not
[surfel_mapping:cmake]   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
[surfel_mapping:cmake]   an ALIAS target is missing?                            
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake]                                                          
[surfel_mapping:cmake] CMake Error at /home/lvsijie/catkin_ws/src/SuMa/CMakeLists.txt:175 (add_executable):
[surfel_mapping:cmake]   Target "genvideo" links to target "Boost::regex" but the target was not
[surfel_mapping:cmake]   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
[surfel_mapping:cmake]   an ALIAS target is missing? 

and my OPENGL version is:
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.5.0 NVIDIA 415.27
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 415.27
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 415.27
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix, 

What's the problem？Could you give me some advice?
Thank you.

jbehley, 2019-02-28:
It seems that boost is the problem. You have to ensure that libboost (filesystem, system, and literal_cast) is available.

Let me know if the problem persist.

Which Ubuntu version are you using?

Lyusijie, 2019-02-28:
> It seems that boost is the problem. You have to ensure that libboost (filesystem, system, and literal_cast) is available.
> 
> Let me know if the problem persist.
> 
> Which Ubuntu version are you using?

Thanks for your reply. My Ubuntu version is 16.04
I have installed libboost-all-dev, should i reinstall it? i am new to ubuntu, so i'm not sure how should i do for this issue :(

jbehley, 2019-02-28:
Hmmm. Never encountered such error. So currently I‘m also clueless. I will have to have a look what the error specifically means. 

Sorry, I will have a look.

jbehley, 2019-03-01:
You should now try to pull again and try to compile. @tano297 run into similar problems and luckily fixed them.

Let me know if you still have problems compiling.

closed 2019-03-01 by jbehley

Lyusijie, 2019-03-05:
> You should now try to pull again and try to compile. @tano297 run into similar problems and luckily fixed them.
> 
> Let me know if you still have problems compiling.

Thank you, it goes well.

commit ed271c2b (2019-03-01): Merge pull request #10 from PRBonn/boost_fix

Fixes #9