jbehley/SuMa #13: How to exploit created map ?

jediofgever, 2019-04-04:
Hi , 
Thanks for excellent work. I have created a a map using my data, now I would like to port the created map, where is this map stored at and what is the extension like, .pcd or .bin ?. I would like to visualize the map with PCL, is there any built in function that can give me map ? 

Thank you for your time

jbehley, 2019-04-04:
Hi @jediofgever,

there is unfortunately now direct way to get the map. Depending on what you need, there are two options:

1. If you need only the aligned point clouds and maybe a voxelization, you can simply take the poses from `SurfelMapping::getOptimizedPoses()` and do the processing yourself. Something like: https://gist.github.com/jbehley/bdc7792ce84cff6233bf6adb30b8be4c
2. If your are interested in the actual surfels, then you have to get the submaps and the current surfels from `SurfelMap`, where the surfels of the active map are stored in `surfels_` and you can use the `get` method of `GlBuffer` to retrieve them. The surfels in the `submapCache_` are then the surfels that are currently in the inactive parts of the surfel map. The` timestamp` of the `Surfel` tells you then the pose `T` which you have to use to get the global position `T * (x,y,z)` of the surfel.

The second option needs some coding and unfortunately I cannot provide right now any assistance with that and there are also no plans to include this feature in future.

jbehley, 2020-12-29:
There seems not to be much activity here. Therefore I close this now. If you have further questions, feel free to reopen the issue again.

closed 2020-12-29 by jbehley