jbehley/SuMa #14: Do you plan to open the "Point Labeler" in the future.

HTLife, 2019-04-10:
In the paper 
>Behley, Jens, et al. "A Dataset for Semantic Segmentation of Point Cloud Sequences." arXiv preprint arXiv:1904.01416 (2019).
you provide an attractive tool(Figure 6) for labeling large point cloud.  Do you plan to open this tool for people who are interested in extending this dataset?

[image]

amiltonwong, 2019-04-10:
:) I'm also waiting for it :P

jbehley, 2019-04-10:
Thanks for the interest in our paper. We plan to publish everything in late summer, i.e., July or August 2019. We also plan to publish our labeling tool as stated in the paper. Stay tuned.

closed 2019-04-10 by jbehley