jbehley/SuMa #15: Docker image for SuMa

HTLife, 2019-04-12:
Hi @jbehley ,
I just create a docker image for SuMa, which could help people quickly evaluate this project.

https://cloud.docker.com/u/tseanliu/repository/docker/tseanliu/suma

The Dockerfile source:
https://github.com/HTLife/suma_docker

jbehley, 2020-12-29:
Sorry that I did not respond (a long time ago). Thanks for your contribution.

I'm not sure what to do with this. If this is still working, I could add this to the readme with the link. Let me know what you think.

 (As it's not a real issue, I simply close this for now.)

closed 2020-12-29 by jbehley

Barnonewdm, 2022-12-28:
just wonder whether you could open the GUI in the container.

> Hi @jbehley , I just create a docker image for SuMa, which could help people quickly evaluate this project.
> 
> https://cloud.docker.com/u/tseanliu/repository/docker/tseanliu/suma
> 
> The Dockerfile source: https://github.com/HTLife/suma_docker

jbehley, 2022-12-28:
I think yes, but didn't try it. It might be necessary to ensure that there is a way for the process in the docker container to use the GPU, but with something like nvidia-docker it's even possible to use CUDA from the container. But I'm not an Docker expert.

Barnonewdm, 2022-12-28:
> I think yes, but didn't try it. It might be necessary to ensure that there is a way for the process in the docker container to use the GPU, but with something like nvidia-docker it's even possible to use CUDA from the container. But I'm not an Docker expert.

Thanks for your fast response. Is there a way to preprocess the data via command without opening the GUI?

jbehley, 2022-12-28:
the code relies on opengl shaders for computation and therefore you need an opengl context. In the experimental ROS node (https://github.com/jbehley/suma_ros), I used this to have an Offscreen Canvas and then call the process frame etc. Important is only that the OpenGL context is in the same thread as the code to run the processing.