jbehley/SuMa #16: How to save poses?

Lyusijie, 2019-04-26:
Hello, i want to get poses of every timestamp. How to use the "Save poses..." button? 
There is nothing When i click it, thank you.

jbehley, 2019-04-27:
Hi @Lyusijie,

you have to enter a filename in the file dialog that should open. However, the default value was not very useful, therefore I changed it to a (hopefully) more sensible value.

Hope that solves your problem.

Lyusijie, 2019-04-27:
Thank you, but there isn't file dialog opened when i click "Save poses..." button.
But "select laserscan file" and "select output directory" is ok

jbehley, 2019-04-27:
This following code connects the button with the `savePoses()` function:
https://github.com/jbehley/SuMa/blob/1beb131cbad33edf157d0a2333fb6193ffd6cc23/src/visualizer/VisualizerWindow.cpp#L172

Unfortunately, it's not possible for me to reproduce your error...

jbehley, 2019-09-06:
@Lyusijie, if the problem still is relevant, please re-open the issue again. Thanks

closed 2019-09-06 by jbehley