jbehley/SuMa #17: The visualizer crashed when click start button, something wrong with opengl

wwtinwhu, 2019-07-18:
Hi Behley, Could you help me with this problem? Thank you very much! Following is what in the terminal when I open visualizer and open one bin file and last click the start button:

./visualizer 
OpenGL Context Version 4.3 core profile
GLEW initialized.
OpenGL context version: 4.3
OpenGL vendor string  : nouveau
OpenGL renderer string: NV117
Extracting surfel maps partially.
Performing frame-to-model matching.
Gtk-Message: 15:12:49.587: GtkDialog mapped without a transient parent. This is discouraged.
calibration filename: /media/weitongwu/Elements/Data/kitti/00/calib.txt...loaded.
ground truth filename: /media/weitongwu/Elements/Data/poses/00.txt
4541 poses read.
Performing frame-to-model matching.
OpenGL Error: invalid value(1281)
In: /home/weitongwu/suma_ws/src/glow/src/glow/GlFramebuffer.cpp on Line: 25
terminate called after throwing an instance of 'std::runtime_error'
  what():  OpenGL error detected.
已放弃 (核心已转储)

jbehley, 2019-07-18:
It seems like that the nouveau driver does not implement all features that I use. Since Nvidia provides more then the core profile functionality with it's driver it happens sometimes that I use more than the required parts of the core profile. Thus, it is very hard for me to really test that I use only core parts of the OpenGL specification.

The easiest way to resolve this problem is to just use the Nvidia driver (if possible). If this is not possible, you could run the program with "gdb" and just provide me with the stack trace to find out where this problem happens.

wwtinwhu, 2019-07-30:
@jbehley  I am very sorry to reply you so late. I will try to use the Nvidia driver later and then if work I will close this issue, thank you very much!

LiShuaixin, 2019-09-05:
@jbehley Hi, I blacklisted the nouveau driver, used Nvidia driver and rebuilt everthing, but the issue was not solved. Would you give me some instruction on which file should I run when I debug the code? I'll feedback you with the stack trace as soon as possible

LiShuaixin, 2019-09-05:
@jbehley Hi, I blacklisted the nouveau driver, used Nvidia driver and rebuilt everthing, but the issue was not solved. Would you give me some instruction on how to debug the visulizer? I'll feedback you with the stack trace as soon as possible

LiShuaixin, 2019-09-05:
The issue is caused by Gtsam. I met the same issue and debug the code.  The back trace showed the issue was occurred when calling function in Gtsam.  The only thing you need to do is recompiling Gtsam of version 4-alpha and then recompiling SuMa

wwtinwhu, 2019-09-06:
> The issue was caused by the wrong version of Gtsam. I met the same issue and I debugged the code to find out where the problem happens. The back trace showed the issue was occurred when calling gtsam function. Now, the only thing you need to do is recompiling gtsam of version[ 4.0.0-alpha2](https://github.com/borglab/gtsam/tree/4.0.0-alpha2) and then recompiling SuMa

@LiShuaixin Thanks for your advice. Recently, I am busy with something else. Then, I will try your method. Best Regards!

jbehley, 2019-09-06:
@LiShuaixin Also thanks from me and I will then close the issue.

If the problem persists, please re-open the issue again.

closed 2019-09-06 by jbehley

zorosmith, 2020-02-28:
Hello @LiShuaixin .
You solve my problem!
Could you tell me how did you figure out this problem?

xdtzzz, 2021-11-08:
hi,@zorosmith，what version number is your computer, i use 4.0.0-alpha2, but it still happen this problem.what should i do?

jbehley, 2021-11-08:
I have it running with ubuntu 20.04, but it also compile and worked with Ubuntu 18.04. important is to compile gtsam with "use_system_eigen" option enabled. Usually, you will get some error in some matrix computation if you did not compile with the system's Eigen version.

xdtzzz, 2021-11-08:
thanks reply so quickly,　I wonder what eigen version corresponding  gtsam4.0.0-alpha2?  In my computer, I can run SuMa++ successfully, but I can't run SuMa