jbehley/SuMa #18: Is it possible to add a ros wrapper?

LiShuaixin, 2019-09-27:
HI @jbehley , thanks a lot for sharing this great work! I tested it using KITTI. The visualization of point cloud  result is good. 

Then I tried to write a ros warper for SuMa, so as to test it on real robots/cars with a 3D lidar. Just like what is used in the qt wrapper `/visualizer/VisualizerWindow.cpp` and implementation `/visualizer/visualizer.cpp`, I create a class `SuMaRosWarper` with a member of `std::shared_ptr<SurfelMapping> fusion_`,  then created a `std::shared_ptr<SurfelMapping>` pointer to initialize it. However, a segment fault was thrown in the construction of `SurfelMapping`. I backtraced the code and found the error occurred when called `glGenBuffers(1, &id_);`:

[image]

BUT I can run your qt version smoothly. What's wrong with this? any suggestion?

jbehley, 2019-09-27:
Since SuMa uses OpenGL for computation, you always need a opengl context in the same (!) thread. Updates and everything must also happen in this thread.

With ROS you have to take into account that nodes might not run on the same thread.

See for instance `glow::X11OffscreenContext` for a headless/windowless context.

But with this you still need a X server running. A true X free context you could get with EGL.

LiShuaixin, 2019-09-27:
Thanks a lot!

closed 2019-09-27 by LiShuaixin

LiShuaixin, 2019-09-27:
Hi, @jbehley  Do you think it's possible to use ros rviz to create a opengl context for SuMa?

jbehley, 2019-09-27:
If you manage to run it in the same thread, then it could work. But I never
tried this. So no guarantees.

MohamedAfifii, 2019-12-09:
@LiShuaixin Were you able to use rviz with suma?