jbehley/SuMa #19: OpenGL renderer string: llvmpipe (LLVM 8.0, 256 bits)

qiaozhijian, 2019-11-25:
Hi Behley, Could you help me with this problem? Thank you very much!  
I run this command
./bin/visualizer ./SuMa/config/default.xml 
Terminal output is following  :

OpenGL Context Version 3.3 core profile
GLEW initialized.
OpenGL context version: 3.3
OpenGL vendor string  : VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 8.0, 256 bits)

jbehley, 2019-12-06:
Hi @qiaozhijian,

sorry for the delay, but I was quite busy with some other stuff.

I would suggest that you run the code without a virtual machine with a real graphics driver. I don't know if the VMWare OpenGL implementation supports all operations needed by our code. It might be that I used some operation that is not strictly supported by OpenGL 3.3 Core Profile. Therefore, I would suggest that you have a real NVIDIA or AMD or Intel GPU driver, since they usally support at least OpenGL 4.0 and usally also more then the Core Profile asks for.

qiaozhijian, 2019-12-06:
> Hi @qiaozhijian,
>
> sorry for the delay, but I was quite busy with some other stuff.
>
> I would suggest that you run the code without a virtual machine with a real graphics driver. I don't know if the [...]

Thank you. But I don't know why it outputs "OpenGL vendor string : VMware, Inc." while my device is equipped with GTX1070. I will try to work it out again. Maybe there are some problems in my Ubuntu.

qiaozhijian, 2019-12-06:
> Hi @qiaozhijian,
>
> sorry for the delay, but I was quite busy with some other stuff.
>
> I would suggest that you run the code without a virtual machine with a real graphics driver. I don't know if the [...]

I have solved the problem by reinstall nvidia-drive. Thank you!
But the new problem ocurrs when I open a ".bin" file

OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1070/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
Gtk-Message: 21:24:21.687: GtkDialog mapped without a transient parent. This is discouraged.
calibration filename: /media/qzj/Document/grow/slam/slamDataSet/kitti/data_odometry_velodyne/dataset/00/calib.txt...段错误 (核心已转储)

jbehley, 2019-12-06:
You need to download the extra calibration files containing a line starting  with`Tr` that is available as extra download on the odometry benchmark website.

qiaozhijian, 2019-12-07:
> You need to download the extra calibration files containing a line starting with`Tr` that is available as extra download on the odometry benchmark website.

OK, thank you~!

closed 2019-12-07 by qiaozhijian

tym2103, 2020-06-09:
> I have solved the problem by reinstall nvidia-drive. Thank you!
> But the new problem ocurrs when I open a ".bin" file
>
> OpenGL Context Version 4.6 core profile
> GLEW initialized.
> OpenGL context [...]

tym2103, 2020-06-09:
How do you solve this problem?Just reinstall nvidia-drive?
I meet the same problem sa you

jbehley, 2020-06-09:
Could you please open a new issue. In this issue, three different things are discussed. And its hard to help, when you don't know what's exactly the problem. However, as a summary for troubleshooting :

1. We mainly tested our approach with Nvidia GPUs and drivers provided by Nvidia under Ubuntu 16.04/18.04. Thus, ensure that you have current Nvidia drivers.
2. We tried to follow the specification of the OpenGL Core Profile and use only functionality described as required, since we made the experience that Intel, Mesa, and ATI often only support the required specifications. However, sometimes there are driver-specifics that we could not test. Please, report such problems and we try to work this out. (Nvidia has a rather feature complete driver and often implements all texture formats, etc. We even experienced that uncompiled shaders got linked by silently compiling them. Very convenient. ;))
3. The KITTI calibration file contained with the images misses the line `Tr:` and must be downloaded separately from the KITTI webpage.

Hope that helps.