jbehley/SuMa #20: Segmentation fault

Sun1307010608, 2019-12-09:
hello, when i compiled this project over, i tried to run ./visualizer in bin fold, the error happend:
'''
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1050 Ti/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
Segmentation fault
'''
any one no why this happended, and how to do to solve this

jbehley, 2019-12-09:
At least your OpenGL configuration seems fine.

How do you call the program? Just `./visualizer` or do you provide some path?

What you can try is to run `gdb ./visualizer`, which will run the debugger and show you were the problem occurs. If you provide me with the information from the backtrack (type `bt` in gdb), I will try to diagnose the problem.

Sun1307010608, 2019-12-09:
@jbehley thank you for your reply, I just ./visualizer ,and not provide any path, when i run  gdb ./visualizer, it shows as follows:
sun@sun-Z370-HD3:~/catkin_ws/src/SuMa/bin$ gdb ./visualizer 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.
(gdb)

jbehley, 2019-12-09:
Ah, sorry. You have to type `run` or `r` to start the debugger.

See also https://cs.baylor.edu/~donahoo/tools/gdb/tutorial.html

Sun1307010608, 2019-12-09:
thanks for your reply, i typed run, it show that :
(gdb) run
Starting program: /home/sun/catkin_ws/src/SuMa/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8795700 (LWP 6520)]
[New Thread 0x7fffddec8700 (LWP 6521)]
[New Thread 0x7fffdd6c7700 (LWP 6522)]
[New Thread 0x7fffdcec6700 (LWP 6523)]
[New Thread 0x7fffcb938700 (LWP 6533)]
OpenGL Context Version 4.5 core profile
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1050 Ti/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::Matrix<float, 4, 4, 0, 4, 4>, Eigen::internal::assign_op<float> > (
    func=..., src=..., dst=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:653
653	  dense_assignment_loop<Kernel>::run(kernel);
(gdb) 

and the Visual interface emerged, and it turns gray, i can not operate any components

jbehley, 2019-12-09:
Problems with Eigen usually have something to do with gtsam and the version of Eigen intalled.

You have to recompile gtsam library with the option `GTSAM_USE_SYSTEM_EIGEN`.

And that the window "turns gray" is normal due to the degbugger that stops the process.

Sun1307010608, 2019-12-09:
thanks, do you mean set GTSAM_USE_SYSTEM_EIGEN on when compile gtsam? I did it already, i remeber that when i not set it on, it occurs an error, and only when i set it on, it comiles ok

nachovizzo, 2019-12-09:
@Sun1307010608 

Please make sure you read through this issue https://github.com/borglab/gtsam/issues/75

- Install Eigen 3.3.7
- Don't use the latest code from gtsam, checkout to this commit https://github.com/borglab/gtsam/commit/c84496632fd6e062e69cb1b9d7b77ac8d9390fbd

Sun1307010608, 2019-12-10:
@nachovizzo thanks for you reply, i have updated my eigen version by 3.3.7, its still occurs segmentation fault. And about gtsam,  this program requires that gtsam >= 4.0, it seems that the latest code from gtsam is 4.0, Do you  mean i should try the 3.x version gtsam,

jbehley, 2019-12-10:
currently does the development branch of gtsam contain a bug. therefore, we suggest to clone and build a specific branch, like `4.0.0`

Sun1307010608, 2019-12-10:
all right, I will change my gtsam version and have a try again.

Sun1307010608, 2019-12-10:
@jbehley thank you very much. ^_^

ProfFan, 2019-12-23:
Hi, Fan from GTSAM here :) 

I think we have fixed some of the bugs related to Eigen in `develop`.

Could you test against the latest `develop` branch of GTSAM and report the results? Thanks a lot! :)

Sun1307010608, 2019-12-28:
孙钦，2942087962@qq.com

Sun1307010608, 2019-12-28:
Hello, Mr. jbehley,
thanks a lot, I take the gtsam version from  'https://github.com/borglab/gtsam/archive/4.0.0-alpha2.zip'  as a replace for my old version, and I successfully setup my project like this:
(but I feel a little confused that how to save the built map and get the localization trajectory after running my dataset, and can you add a rosbag interface for this project, thank you! )

孙钦，2942087962@qq.com

ZuoJiaxing, 2019-12-28:
Yes, I also need a ROS wrapper for this project to run it on our own dataset. Thanks!

Sun1307010608, 2019-12-29:
@ZuoJiaxing Thank you, I am looking forward to your new version of this project. ~_~

ProfFan, 2020-01-01:
> Hi, Fan from GTSAM here :)
> 
> I think we have fixed some of the bugs related to Eigen in `develop`.
> 
> Could you test against the latest `develop` branch of GTSAM and report the results? Thanks a lot! :)

Ping @jbehley :)

jbehley, 2020-01-01:
@ProfFan, thanks for notifying me of the change.

I tried it now and with the current `develop` version it seems to work now. So I can confirm that your changes did resolve the issue. (Maybe @Sun1307010608 can also confirm now that the original issue is now resolved.)

jbehley, 2020-01-01:
@Sun1307010608 and @ZuoJiaxing, I can currently not provide full ROS support. However, I know that people used it inside a node.

I currently just convert bagfiles with a simple node into the `.bin` files. For saving the poses, there is in the menu `File` > `Save Poses...` that writes the poses to a txt file in the KITTI format. However, you can also look inside the corresponding method to access the poses directly.

Please understand that I cannot provide full-fledged support for all possible file formats.

Sun1307010608, 2020-01-02:
@jbehley thanks, maybe I can write a tool to convert the bagfiles to .bin format files as well. but could you tell me how to save the map to my disk.

ProfFan, 2020-01-02:
@jbehley Glad to hear that! Please file a bug report if you have any other issues with GTSAM :)

For all, happy new year!

jbehley, 2020-01-03:
What do you mean with "save the map to my disk", @Sun1307010608?

Do you need the poses? Do you need the surfels? If you want to have an aggregated point cloud given the poses, you can use this [gist](https://gist.github.com/jbehley/bdc7792ce84cff6233bf6adb30b8be4c#file-voxelize-cpp).

But it really depends on what you want.

Sun1307010608, 2020-01-04:
thanks, I mean save the whole point cloud map given the estimated pose by Suma.

jbehley, 2020-01-06:
@Sun1307010608, as mentioned before, there is no direct way to export the map. (Usually, the poses are enough); here a quote from another issue:

> Hi @jediofgever,
> 
> there is, unfortunately, no direct way to get the map. Depending on what you need, there are two options:
> 
>  1. If you need only the aligned point clouds and maybe a voxelization, you can simply take the poses from `SurfelMapping::getOptimizedPoses()` and do the processing yourself. Something like: https://gist.github.com/jbehley/bdc7792ce84cff6233bf6adb30b8be4c
> 
>  2. If you are interested in the actual surfels, then you have to get the submaps and the current surfels from `SurfelMap`, where the surfels of the active map are stored in `surfels_` and you can use the `get` method of `GlBuffer` to retrieve them. The surfels in the `submapCache_` are then the surfels that are currently in the inactive parts of the surfel map. The` timestamp` of the `Surfel` tells you then the pose `T` which you have to use to get the global position `T * (x,y,z)` of the surfel.
> 
> 
> The second option needs some coding and unfortunately I cannot provide right now any assistance with that and there are also no plans to include this feature in future.

Hope that helps.

Sun1307010608, 2020-01-07:
@jbehley 
I got it, thank you!

closed 2020-01-07 by jbehley

marszbz, 2020-03-08:
hi, i tried to run ./visualizer , the error happend:
'''
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
Segmentation fault (core dumped)

'''

 then i tried run gdb ./visualizer, it shows as follows:
"
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.
(gdb) r
Starting program: /home/zbz/generate_pose/src/SuMa/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe902e700 (LWP 4054)]
[New Thread 0x7fffde7e9700 (LWP 4055)]
[New Thread 0x7fffddfe8700 (LWP 4056)]
[New Thread 0x7fffd7fff700 (LWP 4057)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
[New Thread 0x7fffd4e11700 (LWP 4059)]
[New Thread 0x7fffcbfff700 (LWP 4060)]
[New Thread 0x7fffcb7fe700 (LWP 4061)]
[New Thread 0x7fffcaffd700 (LWP 4062)]
[New Thread 0x7fffca7fc700 (LWP 4063)]
[New Thread 0x7fffc9ffb700 (LWP 4064)]
[New Thread 0x7fffc97fa700 (LWP 4065)]
[New Thread 0x7fffc8ff9700 (LWP 4066)]
[New Thread 0x7fffa7fff700 (LWP 4067)]
[New Thread 0x7fffa77fe700 (LWP 4068)]
[New Thread 0x7fffa6ffd700 (LWP 4069)]
[New Thread 0x7fffa67fc700 (LWP 4070)]
Extracting surfel maps partially.
Performing frame-to-model matching.

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
0x00007ffff7b7d357 in _mm256_load_pd (__P=0x555555e1f9b0)
    at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:861
861	  return *(__m256d *)__P;
"

for test opengl,i tried run test_opengl
"
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
[==========] Running 12 tests from 5 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from IcpTest
[ RUN      ] IcpTest.testFrame2Model_opengl
/home/zbz/generate_pose/src/SuMa/test/opengl/icp-test.cpp:39: Failure
      Expected: 2
To be equal to: scans.size()
      Which is: 0
[  FAILED  ] IcpTest.testFrame2Model_opengl (0 ms)
[ RUN      ] IcpTest.testRounding_rect
[       OK ] IcpTest.testRounding_rect (0 ms)
[----------] 2 tests from IcpTest (0 ms total)

[----------] 1 test from PreprocesingTest
[ RUN      ] PreprocesingTest.generateImage
>>> WARNING: NEED TO FIX THIS ISSUE. <<< 
[       OK ] PreprocesingTest.generateImage (1 ms)
[----------] 1 test from PreprocesingTest (1 ms total)

[----------] 2 tests from DepthImageGeneratorTest
[ RUN      ] DepthImageGeneratorTest.quantization
[       OK ] DepthImageGeneratorTest.quantization (4 ms)
[ RUN      ] DepthImageGeneratorTest.testNDC2textureCoords
[       OK ] DepthImageGeneratorTest.testNDC2textureCoords (6 ms)
[----------] 2 tests from DepthImageGeneratorTest (11 ms total)

[----------] 6 tests from NdcTest
[ RUN      ] NdcTest.testFramebuffer_mini
[       OK ] NdcTest.testFramebuffer_mini (1 ms)
[ RUN      ] NdcTest.testFramebuffer_mini_uneven
/home/zbz/generate_pose/src/SuMa/test/opengl/testNDC.cpp:122: Failure
      Expected: int(expected.x)
      Which is: 10
To be equal to: int(out.x)
      Which is: 0
[  FAILED  ] NdcTest.testFramebuffer_mini_uneven (2 ms)
[ RUN      ] NdcTest.testFramebuffer_square
[       OK ] NdcTest.testFramebuffer_square (26 ms)
[ RUN      ] NdcTest.testFramebuffer_rectangle
[       OK ] NdcTest.testFramebuffer_rectangle (12 ms)
[ RUN      ] NdcTest.testFramebuffer_uneven
[       OK ] NdcTest.testFramebuffer_uneven (2 ms)
[ RUN      ] NdcTest.testFramebuffer_uneven_large
[       OK ] NdcTest.testFramebuffer_uneven_large (208 ms)
[----------] 6 tests from NdcTest (251 ms total)

[----------] 1 test from JacobianTest
[ RUN      ] JacobianTest.testjacobian
/home/zbz/generate_pose/src/SuMa/test/opengl/jacobian-test.cpp:58: Failure
      Expected: 2
To be equal to: scans.size()
      Which is: 0
[  FAILED  ] JacobianTest.testjacobian (0 ms)
[----------] 1 test from JacobianTest (0 ms total)

[----------] Global test environment tear-down
[==========] 12 tests from 5 test cases ran. (263 ms total)
[  PASSED  ] 9 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] IcpTest.testFrame2Model_opengl
[  FAILED  ] NdcTest.testFramebuffer_mini_uneven
[  FAILED  ] JacobianTest.testjacobian
"
any one know why this happended, and how to do to solve this

marszbz, 2020-03-08:
hi,when i compiled this project over, i tried to run ./visualizer , the error happend:

```
_OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
Segmentation fault (core dumped)_
```
then i tried run gdb ./visualizer, it shows as follows:

```
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later http://gnu.org/licenses/gpl.html
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
http://www.gnu.org/software/gdb/bugs/.
Find the GDB manual and other documentation resources online at:
http://www.gnu.org/software/gdb/documentation/.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.
(gdb) r
Starting program: /home/zbz/generate_pose/src/SuMa/bin/visualizer
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe902e700 (LWP 4054)]
[New Thread 0x7fffde7e9700 (LWP 4055)]
[New Thread 0x7fffddfe8700 (LWP 4056)]
[New Thread 0x7fffd7fff700 (LWP 4057)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
[New Thread 0x7fffd4e11700 (LWP 4059)]
[New Thread 0x7fffcbfff700 (LWP 4060)]
[New Thread 0x7fffcb7fe700 (LWP 4061)]
[New Thread 0x7fffcaffd700 (LWP 4062)]
[New Thread 0x7fffca7fc700 (LWP 4063)]
[New Thread 0x7fffc9ffb700 (LWP 4064)]
[New Thread 0x7fffc97fa700 (LWP 4065)]
[New Thread 0x7fffc8ff9700 (LWP 4066)]
[New Thread 0x7fffa7fff700 (LWP 4067)]
[New Thread 0x7fffa77fe700 (LWP 4068)]
[New Thread 0x7fffa6ffd700 (LWP 4069)]
[New Thread 0x7fffa67fc700 (LWP 4070)]
Extracting surfel maps partially.
Performing frame-to-model matching.

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
0x00007ffff7b7d357 in _mm256_load_pd (__P=0x555555e1f9b0)
at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:861
861 return *(__m256d *)__P;
"

```
for test opengl,i tried run test_opengl

```
GLEW initialized.
OpenGL context version: 4.5
OpenGL vendor string : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
[==========] Running 12 tests from 5 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from IcpTest
[ RUN ] IcpTest.testFrame2Model_opengl
/home/zbz/generate_pose/src/SuMa/test/opengl/icp-test.cpp:39: Failure
Expected: 2
To be equal to: scans.size()
Which is: 0
[ FAILED ] IcpTest.testFrame2Model_opengl (0 ms)
[ RUN ] IcpTest.testRounding_rect
[ OK ] IcpTest.testRounding_rect (0 ms)
[----------] 2 tests from IcpTest (0 ms total)

[----------] 1 test from PreprocesingTest
[ RUN ] PreprocesingTest.generateImage

WARNING: NEED TO FIX THIS ISSUE. <<<
[ OK ] PreprocesingTest.generateImage (1 ms)
[----------] 1 test from PreprocesingTest (1 ms total)

[----------] 2 tests from DepthImageGeneratorTest
[ RUN ] DepthImageGeneratorTest.quantization
[ OK ] DepthImageGeneratorTest.quantization (4 ms)
[ RUN ] DepthImageGeneratorTest.testNDC2textureCoords
[ OK ] DepthImageGeneratorTest.testNDC2textureCoords (6 ms)
[----------] 2 tests from DepthImageGeneratorTest (11 ms total)

[----------] 6 tests from NdcTest
[ RUN ] NdcTest.testFramebuffer_mini
[ OK ] NdcTest.testFramebuffer_mini (1 ms)
[ RUN ] NdcTest.testFramebuffer_mini_uneven
/home/zbz/generate_pose/src/SuMa/test/opengl/testNDC.cpp:122: Failure
Expected: int(expected.x)
Which is: 10
To be equal to: int(out.x)
Which is: 0
[ FAILED ] NdcTest.testFramebuffer_mini_uneven (2 ms)
[ RUN ] NdcTest.testFramebuffer_square
[ OK ] NdcTest.testFramebuffer_square (26 ms)
[ RUN ] NdcTest.testFramebuffer_rectangle
[ OK ] NdcTest.testFramebuffer_rectangle (12 ms)
[ RUN ] NdcTest.testFramebuffer_uneven
[ OK ] NdcTest.testFramebuffer_uneven (2 ms)
[ RUN ] NdcTest.testFramebuffer_uneven_large
[ OK ] NdcTest.testFramebuffer_uneven_large (208 ms)
[----------] 6 tests from NdcTest (251 ms total)

[----------] 1 test from JacobianTest
[ RUN ] JacobianTest.testjacobian
/home/zbz/generate_pose/src/SuMa/test/opengl/jacobian-test.cpp:58: Failure
Expected: 2
To be equal to: scans.size()
Which is: 0
[ FAILED ] JacobianTest.testjacobian (0 ms)
[----------] 1 test from JacobianTest (0 ms total)

[----------] Global test environment tear-down
[==========] 12 tests from 5 test cases ran. (263 ms total)
[ PASSED ] 9 tests.
[ FAILED ] 3 tests, listed below:
[ FAILED ] IcpTest.testFrame2Model_opengl
[ FAILED ] NdcTest.testFramebuffer_mini_uneven
[ FAILED ] JacobianTest.testjacobian
```
anyone know why this happended, and how to solve this

jbehley, 2020-03-25:
@Sun1307010608 It might be a problem related to different Eigen versions of gtsam and your system's Eigen. Therefore it usually helps to either install gtsam with the option  `USE_SYSTEM_EIGEN` or installing a specific version of Eigen. (See also this [issue]( https://github.com/PRBonn/semantic_suma/issues/14))

hope that resolves your problems.

xdtzzz, 2021-11-08:
hi, @jbehley, thanks for your code!!!
I meet the same question, and I use 'gdb ./visualizer' and input 'r' ,  it shows these:
Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (
    this=0x200000002) at /usr/include/c++/7/bits/shared_ptr_base.h:151
151		if (__gnu_cxx::__exchange_and_add_dispatch(&_M_use_count, -1) == 1)

could you tell me what problem is it?
Best wishes!!!

xdtzzz, 2021-11-08:
and when I input 'bt', it shows 'No stack'. @jbehley