jbehley/SuMa #22: CUDA porting

ceccocats, 2020-05-12:
Hi,
this project is awesome! thank you very much for sharing your work.
I was thinking to port this project to CUDA, i know that openGL is much more portable but maybe in CUDA it will gain in performances.
I though that even before starting to work is better to ask the author :)
Do you think the porting is possible?
Do you think it will gain some performance?

thanks,
Francesco

jbehley, 2020-05-12:
Hi @ceccocats,

it's definitely portable and for some parts I would expect also to have some performance gains, since I had sometimes to use some "hacks" to get OpenGL to do what I want. Thus, for some parts, you could get a more sane implementation (storing matrices in textures with accessing them with texture coordinates is definitely not the nicest thing...)

However, you might still want to use the rendering capabilities of OpenGL or Vulkan (which would be my choice today if I had more time.). Otherwise you have to write the z-buffering, rasterization, etc. yourself, which would be inferior to using the GPU for it's original purpose (graphics). But I would try to use for surfel rendering now instancing instead of a geometry shader for rendering the surfels.

Let me know when you have a runtime comparison.

jbehley, 2020-12-29:
I close this issue now. if you have any updates or questions regarding this, please simply reopen the issue.

closed 2020-12-29 by jbehley