jbehley/SuMa #24: How to use the simulation_reader ?

Gatsby23, 2020-05-14:
Hi, I have noticed that the io files have the simulation_reader about the SLAM tests in simulation environment. But I don't know which simulation platform can provide these simulation data, could you give me some hint ？

jbehley, 2020-05-15:
This part was related to a custom very basic simulation of a LiDAR sensor. I could upload an example file if needed. But this is not very well maintained code. Thus expect some rough edges.

Gatsby23, 2020-05-15:
> This part was related to a custom very basic simulation of a LiDAR sensor. I could upload an example file if needed. But this is not very well maintained code. Thus expect some rough edges.

OK, I wanna to see the example, thank you at all. By the way, do you use the airsim or carla to do
 the simulation ?

closed 2020-07-08 by Gatsby23