jbehley/SuMa #25: Something about the GlShader::fromCache();

Gatsby23, 2020-07-08:
I'm sorry, I wondering how did the GlShaderCache initialize and where did the sources_ filled with the GLSL have provide？

jbehley, 2020-07-14:
Please see my answer at https://github.com/jbehley/glow/issues/9#issuecomment-658401197

Just for convenience:
> That's a good question and I'm still happy about the solution, even if it is a little bit of magic involved.
> 
> First, the cmake command COMPILE_SHADER takes the code in the text file and copies it into a class `CacheInitializer` (see `visualization_shader.cpp` or `computation_shaders.cpp` in the build folder). There the constructor gets an instance of the `GlShaderCache` and adds the corresponding shader sources from the COMPILE_SHADER macro.
> 
> Now comes the magic: the `CacheInitializer` is used as a static variable: `static CacheInitalizer __init__;` inside the same cpp file. :tada:
> 
> The static variable will be initialized as you define it using the constructor. Thus it will add the shader sources to the cache, et voila. The shaders are added without explicitly calling ever some initialization code!
> 
> You just have to now link the cpp into the binary and that's it. Real magic! :tophat: :sparkle:

Gatsby23, 2020-08-06:
> Please see my answer at [jbehley/glow#9 (comment)](https://github.com/jbehley/glow/issues/9#issuecomment-658401197)
> 
> Just for convenience:
> 
> > That's a good question and I'm still happy about the solution, even if it is a little bit of magic involved.
> > First, the cmake command COMPILE_SHADER takes the code in the text file and copies it into a class `CacheInitializer` (see `visualization_shader.cpp` or `computation_shaders.cpp` in the build folder). There the constructor gets an instance of the `GlShaderCache` and adds the corresponding shader sources from the COMPILE_SHADER macro.
> > Now comes the magic: the `CacheInitializer` is used as a static variable: `static CacheInitalizer __init__;` inside the same cpp file. 🎉
> > The static variable will be initialized as you define it using the constructor. Thus it will add the shader sources to the cache, et voila. The shaders are added without explicitly calling ever some initialization code!
> > You just have to now link the cpp into the binary and that's it. Real magic! 🎩 ❇️

Yes! I already find it, thank you very much! SuMa is a really great program!

closed 2020-08-06 by Gatsby23