jbehley/SuMa #26: Visualization raises a segment error

zhoupengwei, 2020-07-31:
Hi!
Thanks for excellent work. When I finished compiling and running the visual executable, a segment error was triggered. 
I checked the source code, the problem may appear new operator. 
The code that causes the error is
std::shared_ptr<SurfelMapping> fusion = std::shared_ptr<SurfelMapping>(new SurfelMapping(params));
which in visualizer.cpp. 
the error is that
zpw@msi:~/slam/suma_ws/src/SuMa/bin$ ./visualizer 
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce RTX 2060/PCIe/SSE2
段错误 (核心已转储)
Do you have this error when you run the code? Can you give some solutions?

Thanks

jbehley, 2020-07-31:
Can you provide the backtrace(bt) from gdb? Run `gdb -ex run ./visualizer`and then type `bt` when the program crashes.

Bur my suspicion is that it's an problem with Eigen and latest gtsam. For gtsam specific version work, please see: https://github.com/jbehley/SuMa/issues/20

zhoupengwei, 2020-08-01:
@jbehley, yes,I chang the gtsam version by 4.0.0-alpha2 and eigen downgrade at 3.2.7 , now it worked well,thank you very much.

HIT-CRT, 2020-08-04:
I had the same problem .I solved it by add 'EIGEN_MAKE_ALIGNED_OPERATOR_NEW'  in  file 'SurfelMapping.h'.
"If you define a structure having members of fixed-size vectorizable Eigen types, you must overload its "operator new" so that it generates 16-bytes-aligned pointers. Fortunately, Eigen provides you with a macro EIGEN_MAKE_ALIGNED_OPERATOR_NEW that does that for you."
'http://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html'

HIT-CRT, 2020-08-04:
I meet the same problem.I solved it by adding 'EIGEN_MAKE_ALIGNED_OPERATOR_NEW' to the header file, and add ' Eigen::aligned_allocator' to all the vector or map whitch use Eigen object.

'http://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html':
"If you define a structure having members of fixed-size vectorizable Eigen types, you must overload its "operator new" so that it generates 16-bytes-aligned pointers. Fortunately, Eigen provides you with a macro EIGEN_MAKE_ALIGNED_OPERATOR_NEW that does that for you."

'http://eigen.tuxfamily.org/dox/group__TopicStlContainers.html':
'Using STL containers on fixed-size vectorizable Eigen types, or classes having members of such types, requires taking the following two steps:
*A 16-byte-aligned allocator must be used. Eigen does provide one ready for use: aligned_allocator. 
*If you want to use the std::vector container, you need to #include <Eigen/StdVector>.
These issues arise only with fixed-size vectorizable Eigen types and structures having such Eigen objects as member. For other Eigen types, such as Vector3f or MatrixXd, no special care is needed when using STL containers.
'

Rogers34, 2020-08-19:
@zhoupengwei hi, I use ubuntu 18.04.5, nvidia-drivers-450(GeForce GTX 1660Ti ), gtsam4.0.0-alpha2 eigen3.2.7
but i receive Segmentation fault.
Could you tell me which distribution of your system?  And which vesion of the lib i use is wrong.

zhoupengwei, 2020-08-19:
I am using the same version in my system, You need to specify to use Eigen
in the system when cmake is compiled.
There are some main notices that are due to source installation of Eigen,
its path will be located at /usr/local/include/.
The path to install Eigen directly using the command is different, it is
located at /usr/include.
I just copy  /usr/local/include/eigen3 to /usr/include/ after finished
compiling it.
The compilation command is roughly as follows:
cd gtsam/
mkdir build && cd build
cmake -DGTSAM_USE_SYSTEM_EIGEN=ON ..
sudo make install

Hope to help you!
Regards
PengWei

Rogers34, 2020-08-19:
我的eigen是安装在usr/include下的
gtsam编译的时候也选了use_system_eigen
还有没有其他的要注意的

zhoupengwei, 2020-08-19:
没有了, 这个错误是由于Eigen3.3系列的版本引起的
换成3.2系列的就可以了, gtsam可以用最新的版本编译.

Rogers34, 2020-08-20:
我尝试了很多组合，最后gtsam4.0.3+eigen3.2.10 能运行

jbehley, 2020-12-29:
Is the problem resolved?

zpw6106, 2020-12-31:
Yes, this problem has been solved, thank you very much！

jbehley, 2020-12-31:
Good to hear! Then I close now this issue. If you could provide some comment on the actual solution for others that would be highly appreciated.

closed 2020-12-31 by jbehley