jbehley/SuMa #28: OpenGL Error

Rogers34, 2020-08-13:
Hello dear friends!
Thank you for the amazing work.
I have install Ubuntu16.04 on my computer.
I download and compile your project follow the steps in the README.md. 
when i run the ./visualizer i get the information as bellow

OpenGL Context Version 3.3 core profile
GLEW initialized.
OpenGL context version: 3.3
OpenGL vendor string  : VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 6.0, 256 bits)
Extracting surfel maps partially.
OpenGL Error: invalid enumerant(1280)
In: /home/rogers/catkin_ws/src/SuMa/src/core/SurfelMap.cpp on Line: 298
terminate called after throwing an instance of 'std::runtime_error'
  what():  OpenGL error detected.
Aborted (core dumped)

I don't know what is the problem. Could you help me find this problem out? Thank you very much.

jbehley, 2020-08-13:
Could you try to run the program with a native Nvidia or AMD driver? It seems like that the VM driver does not support all instructions needed to run the visualizer.

Rogers34, 2020-08-14:
I install the opengl use the command as bellow
sudo apt-get install build-essential libgl1-mesa-dev
sudo apt-get install freeglut3-dev
sudo apt-get install libglew-dev libsdl2-dev libsdl2-image-dev libglm-dev libfreetype6-dev
Should I compile and install opengl with source code?

jbehley, 2020-08-14:
You apparently use VMware to have a virtual machine with Linux:

```
OpenGL context version: 3.3
OpenGL vendor string : VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 6.0, 256 bits)
```

Install an Nvidia, AMD or Intel (mesa) graphics driver to execute the program. It has nothing to do with OpenGL. See also https://github.com/jbehley/SuMa/issues/19

Rogers34, 2020-08-19:
@jbehley 

I have installed nvidia driver in ubuntu 18.04.5
I compiled code with eigen3.3.7 and gtsam4.0.3.
I received SIGSEGV
gdb infomation as bellow.

GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.
(gdb) r
Starting program: /home/rogers/catkin_ws/src/SuMa/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe9659700 (LWP 8853)]
[New Thread 0x7fffdee93700 (LWP 8854)]
[New Thread 0x7fffde692700 (LWP 8855)]
[New Thread 0x7fffdc9d7700 (LWP 8856)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
0x00005555555b6d14 in _mm256_store_ps (__A=..., __P=<optimized out>) at /usr/lib/gcc/x86_64-linux-gnu/7/include/avxintrin.h:880
880	  *(__m256 *)__P = __A;

Rogers34, 2020-08-19:
@jbehley 

sometimes it is like this

GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.
r(gdb) r
Starting program: /home/rogers/catkin_ws/src/SuMa/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe9659700 (LWP 9011)]
[New Thread 0x7fffdee93700 (LWP 9012)]
[New Thread 0x7fffde692700 (LWP 9013)]
[New Thread 0x7fffdc9d7700 (LWP 9014)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
ViewportWidget::paintGL (this=0x555555c5dac0) at /home/rogers/catkin_ws/src/SuMa/src/visualizer/ViewportWidget.cpp:541
541	  view_ = camera_.matrix();

Rogers34, 2020-08-20:
finally gtsam4.0.3+eigen3.2.10 works for me.
Thanks a lot for your help!
I have download som data of kitti.
But the .bin data is very small(only 2MB), so the program stoped very soon.
Could you tell where can i download a bigger data, so i can run a whole map?

jbehley, 2020-08-20:
You can download the KITTI Odometry Benchmark. Sequence 00 is a long trajectory with multiple loop closures.

Rogers34, 2020-08-26:
i have downloaded  Sequence 00. It has 4649 2.0MB bins. 
Then I cat *.bin > total.bin
When i use the total.bin, the visualizer killed

OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 Ti/PCIe/SSE2
Extracting surfel maps partially.
Performing frame-to-model matching.
Gtk-Message: 10:09:00.709: GtkDialog mapped without a transient parent. This is discouraged.
calibration filename: /home/rogers/data/calib.txt...not found.
ground truth filename: /home/poses/data.txt
not found.
Killed

Is the bin file too big to run the visualizer?

jbehley, 2020-08-26:
That is definetly to large for your GPU memory.

if you have downloaded the odometry data, select from sequence 00 the velodyne folder. Alternatively you can start the program with `./visualizer ../config/default.xml ~/kitti-odometry/dataset/sequences/00/velodyne/`. If you then increast the number of history scans you can see a longer history. 

[image]

In the defaul.xml are parameters influencing how many scans are buffered and at which stride. stride*size is the maximum length of the history covered. For sequence 0 you might increase this to stride 25, for instance.

```xml
<param name="history size" type="integer">200</param>
<param name="history stride" type="integer">20</param>
```

hope that solves your problem.

Rogers34, 2020-10-19:
Hello dear friends,
    I am trying to use our own data of VLP16 lidar.
    I modified the param as follow

<config>
  
  <param name="data_width" type="integer">765</param>
  <param name="data_height" type="integer">16</param>
  <param name="data_fov_up" type="float">15.0</param>
  <param name="data_fov_down" type="float">-15.0</param>
  <param name="max_depth" type="float">200.0</param>
  <param name="min_depth" type="float">0.0</param>
                      ...
  
  <param name="model_width" type="integer">765</param>
  <param name="model_height" type="integer">16</param>
  <param name="model_fov_up" type="float">15.0</param>
  <param name="model_fov_down" type="float">-15.0</param>
  <param name="model_max_depth" type="float">200.0</param>
  <param name="model_min_depth" type="float">0.0</param>
                       ...
</config>

but i get the unexpected result like this 

[image]
[image]

The data was collected in a indoor environment.
Could you give me some advices to solve this problem? Thanks a lot!

jbehley, 2020-10-19:
It appears that ground points are missing, but these are needed to fix the pitch angle. Without ground points it's unlikely to get a good result with a 6D pose estimate as it cannot fix this rotation. If you still want to make mapping in the plane, a 2D mapping approach like cartographer (https://github.com/cartographer-project/cartographer) 

PS: Since you are indors, ceiling points would also be fine.

Rogers34, 2020-10-20:
Thanks for your advices.
I am using another data set which was collected in the outdoor and had the ground points.
But it seems that the result is not correct as well.
[image]
[image]
Shoud i modify the icp params?

Rogers34, 2020-10-20:
Some output message in the terminal :

Gtk-Message: 14:16:57.188: GtkDialog mapped without a transient parent. This is discouraged.
calibration filename: /home/rogers/Data/calib.txt...not found.
ground truth filename: /home/poses/Data.txt
not found.
t_err: 0.15531, r_err: 0.148142
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.
t_err: 0.255079, r_err: 0.252976
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.
t_err: 0.90735, r_err: 0.179166
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.
t_err: 3.31942, r_err: 0.198348
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.
t_err: 1.1902, r_err: 0.227667
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.
t_err: 2.58965, r_err: 0.157356
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.
t_err: 1.22369, r_err: 0.265477
Lost track: Pose increment change too large. Falling back to frame-to-frame tracking.

jbehley, 2020-10-20:
Yes, you have to find the right parameters, since our values are adapted to the KITTI odometry dataset with a fast moving car.

Please see https://github.com/jbehley/SuMa/blob/master/config/vlp16_example.xml for an example configuration that provided good results. The width `model_width` and `data_width` seems to be to large. For slowly moving robots is seems to be also help to reduce the `icp-max-distance`.

That the approach fallsback to the frame-to-frame tracking is usually not a good sign.

Rogers34, 2020-10-20:
I tried the vlp16_example.xml, but the result was also not  good. 
In the data set, there are 10906.1(mean) points in one frame. 10906.1/16=681.633, so i modified the width to 681.
And i decreased the icp-max-distance from 2.0 to 1.0, icp-max-angle from  30 to 10. After that, result was still not good.

Rogers34, 2020-10-20:
I read the pointcloud from the pcap file, than save them to the bin file.
Is there any problems with our dataset?

Rogers34, 2020-10-20:
Maybe the problem is the name of the bin file.
I fixed it from 1.bin to 000001.bin.
Now it seem that the result is better.
But there is an other problem now.
The real trajectory of the dataset is a loop, but the result of the program is not.
[image]
[image]
Could you give some advices to solve this problem? Thanks a lot!!

jbehley, 2020-10-20:
In the best case it should find a loop closure and correct the drift. However, there are multiple parameters that influence this and these have to be adjusted for the velodyne VLP-16 since it covers more from the tree canopy and such things, it will also have a higher residual and the thresholds are not completely right.

When you go to "loop closure" you can actually switch on that he shows the loop closures candidates. if it finds reasonable candidates but does not include a closure, it means that the threshold is just to hight. adjust the values.

I can unfortunately not give you other then the advice to try out values. we are working hard on better solutions, but this is now the state of SuMa.

jbehley, 2020-12-29:
The problem seems resolved. If you have further comments, please reopen the issue.

closed 2020-12-29 by jbehley