jbehley/SuMa #29: Segmentation fault

jhjun724, 2020-11-11:
Hello.

I'm trying to run ./visualizer but there was "Segmentation fault".
I think I tried to solve this problem with related issues like #20 and [https://github.com/PRBonn/semantic_suma/issues/14](url).

My Eigen and gtsam versions are 3.3.7 and 4.0.0, respectively.

And I also compiled gtsam with eigen.
`make GTSAM_USE_SYSTEM_EIGEN='on'
make install`

But a similar problem still remains.
`$ gdb ./visualizer 
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.
(gdb) run
Starting program: /home/vdclab/catkin_ws/src/SuMa/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8b73700 (LWP 13785)]
[New Thread 0x7fffde5f4700 (LWP 13787)]
[New Thread 0x7fffdddf3700 (LWP 13788)]
[New Thread 0x7fffd7fff700 (LWP 13789)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 980 Ti/PCIe/SSE2

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
0x0000000000450056 in Eigen::internal::pload<float __vector(8)>(Eigen::internal::unpacket_traits<float __vector(8)>::type const*) (from=<optimized out>)
    at /usr/local/include/eigen3/Eigen/src/Core/arch/AVX/PacketMath.h:214
214	template<> EIGEN_STRONG_INLINE Packet8f pload<Packet8f>(const float*   from) { EIGEN_DEBUG_ALIGNED_LOAD return _mm256_load_ps(from); }`

Should I downgrade my Eigen version to 3.2.1 or 3.2.10 like [https://github.com/PRBonn/semantic_suma/issues/14](url)?
Please let me know how to solve this problem.

Thank you.

jhjun724, 2020-11-11:
Here is my environment.
```
$ glxinfo | grep "version"
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 450.80.02
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 450.80.02
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 450.80.02
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_group_vote, GL_EXT_shader_implicit_conversions, 
```
So I build with below command.
```
$ catkin build --save-config -i --cmake-args -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=460 -DENABLE_NVIDIA_EXT=YES
```
And this is version of Eigen.
```
$ pkg-config --modversion eigen3
3.3.7
```
I don't know how to check my gtsam version.
But I installed 4.0.0.(not alpha2)
```
~/Documents/gtsam-4.0.0$ mkdir build
~/Documents/gtsam-4.0.0$ cd build
~/Documents/gtsam-4.0.0/build$ make GTSAM_USE_SYSTEM_EIGEN='on'
~/Documents/gtsam-4.0.0/build$ make install
```

jbehley, 2020-11-11:
It seems that gtsam with specific versions of Eigen has problems and newer versions of gtsam reintroduce these problems. We can only confirm that version `4.0.0-alpha2` works with newer versions of Eigen. 

Since it worked for other people, I would now suggest to get version `4.0.0-alpha2` and try it with `GTSAM_USE_SYSTEM_EIGEN` activated.

Hope that resolves the issues.

jhjun724, 2020-11-12:
I solved this problem by editing CMakeLists.txt files in SuMa and GTSAM.

I think this problem was related with Eigen libraries.
If I install Eigen by below command,
```
sudo apt install libeigen3-dev
```
the library would be installed in `/usr/include`
and if I install Eigen with .zip or .tar.gz file, the library would be installed in `/usr/local/include`.
(It can be different for each computer.)

In SuMa CMakeLists.txt file, you can find out that SuMa is using Eigen library in `/usr/include`. (line 36)
And you can find `include_directories(AFTER "${EIGEN3_INCLUDE_DIR}")` and `set(GTSAM_EIGEN_INCLUDE_PREFIX "${EIGEN3_INCLUDE_DIR}")` in `/gtsam-4.0.0-alpha2/CMakeLists.txt`. (line 228, 231)

It’s ambiguous whether GTSAM is using Eigen library in `/usr/include` or `/usr/local/include`. I think this made some crashes and caused my problem.
Also I can't determine the version of Eigen if I use the library in `/usr/include`.
So I changed CMakeLists.txt files.

First, `/SuMa/CMakeLists.txt`(line 36)
from
```
include_directories(${QT5_INCLUDE_DIRS} ${GLEW_INCLUDE_DIRS} ${GTSAM_INCLUDE_DIRS} /usr/include/eigen3)
```
to
```
include_directories(${QT5_INCLUDE_DIRS} ${GLEW_INCLUDE_DIRS} ${GTSAM_INCLUDE_DIRS} /usr/local/include/eigen3)
```

Second, before compile GTSAM, `/gtsam-4.0.0-alpha2/CMakeLists.txt`(line 226~231)
from
```
if(GTSAM_USE_SYSTEM_EIGEN)
	find_package(Eigen3 REQUIRED)
	include_directories(AFTER "${EIGEN3_INCLUDE_DIR}")

	# Use generic Eigen include paths e.g. <Eigen/Core>
	 set(GTSAM_EIGEN_INCLUDE_PREFIX "${EIGEN3_INCLUDE_DIR}")
```
to
```
if(GTSAM_USE_SYSTEM_EIGEN)
	find_package(Eigen3 REQUIRED)
	include_directories(AFTER "/usr/local/include/eigen3")

	# Use generic Eigen include paths e.g. <Eigen/Core>
	 set(GTSAM_EIGEN_INCLUDE_PREFIX "/usr/local/include/eigen3")
```
and compile GTSAM with Eigen.

It works for Eigen 3.2.10 and 3.3.7 with this changes.(GTSAM: 4.0.0-alpha2)
(And gtsam-4.0.0-alpha2 requires at least Eigen 3.2.5 if you are using MKL.)

Thank you for your comment and source code jbehley.

closed 2020-11-16 by jhjun724

xdtzzz, 2021-11-08:
Hi, @ jhjun724 , I meet the same problem . I use your method, but it still append this problem, what should i do? 
Best wishes!