jbehley/SuMa #30: Used in stereo camera

liuzhenboo, 2020-12-27:
Can this project be expanded for outdoor reconstruction of binocular cameras？

jbehley, 2020-12-29:
I would guess with some modifications to the projection function, it should be possible. 

The only thing that our algorithm needs is a vertex map and normal map (which is generated from the vertex map) and a projection function to get from three-dimensional points to the image coordinates in these vertex and normal maps.

liuzhenboo, 2021-01-03:
> I would guess with some modifications to the projection function, it should be possible.
>
> The only thing that our algorithm needs is a vertex map and normal map (which is generated from the vertex [...]

Thanks a lot, I will try it!

closed 2021-11-08 by jbehley