jbehley/SuMa #32: segmentation fault

hungkyun, 2021-03-04:
two days ago i successfully built and use for pose estimation,but today it turns to segmentation fault,here's the gdb imformation:
 Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x20) at malloc.c:3113
3113	malloc.c. no such or directory.

i met seg fault for the very beginning and i"get gtsam version 4.0.0-alpha2 and try it with GTSAM_USE_SYSTEM_EIGEN activated."to solve the problem.

hope that somthing wrong can be found.

jbehley, 2021-03-04:
It would maybe help to see the complete backtrack and to get an idea, where the error originates. Currently, I can only deduce that there is an error in freeing memory.

hungkyun, 2021-03-04:
it works after i rebuild the gtsam and suma like before,thank you for helpping!

closed 2021-03-04 by hungkyun