jbehley/SuMa #34: segmentation fault

xdtzzz, 2021-11-09:
hi, thanks for your code!!!

I meet the same question as some issues, and I use 'gdb ./visualizer' and input 'bt' and  'r' , it shows these:

xdt@xdt:~/catkin_ws2/src/SuMa/bin$ gdb ./visualizer
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...done.

(gdb) bt
No stack.

(gdb) r
Starting program: /home/xdt/catkin_ws2/src/SuMa/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeba04700 (LWP 19741)]
[New Thread 0x7fffe0c99700 (LWP 19742)]
[New Thread 0x7fffdbfff700 (LWP 19743)]
[New Thread 0x7fffda785700 (LWP 19744)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1650 Ti/PCIe/SSE2
Extracting surfel maps partially.
Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x203d206e6f697469)
    at /usr/include/c++/7/bits/shared_ptr_base.h:151
151		if (__gnu_cxx::__exchange_and_add_dispatch(&_M_use_count, -1) == 1)

could you please tell me what should I do?
Best wishes!

Jimmij50, 2021-11-09:
same error

jbehley, 2021-11-09:
currently, I have no idea what could be wrong here. usual suspects are Eigen + gtsam, but these usually result in errors with some part of the Eigen libary. Here the problem is with some shared pointer. I just compiled and run the code on Ubuntu 20.04 with gcc 9.3.0. Apart from some issues with boost, it worked just fine.

Therefore, I have currently no idea where the problem could be.

closed 2023-05-10 by jbehley