jbehley/SuMa #38: Error in compilation

Barnonewdm, 2022-12-28:
Any idea of the following error when perform `catkin build --save-config -i --cmake-args -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=430 -DENABLE_NVIDIA_EXT=YES`?

```
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create -rdynamic CMakeFiles/cmTC_3b2ac.dir/CheckFunctionExists.c.o -o cmTC_3b2ac  -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_3b2ac.dir/build.make:98: recipe for target 'cmTC_3b2ac' failed
make[1]: *** [cmTC_3b2ac] Error 1
make[1]: Leaving directory '/workspace/catkin_ws/build/surfel_mapping/CMakeFiles/CMakeTmp'
Makefile:127: recipe for target 'cmTC_3b2ac/fast' failed
make: *** [cmTC_3b2ac/fast] Error 2
```

jbehley, 2022-12-28:
The relevant line in the error is: `/usr/bin/ld: cannot find -lpthreads`

Not sure why it isn't installed with libc, but there are multiple results on google. (It might be sufficient to install `libc6-dev`, but I didn't check it.)

Barnonewdm, 2022-12-28:
> libc6-dev

Thanks, this lib is installed. Therefore, not working.

jbehley, 2022-12-28:
Then I have no idea why the linker doesn't find the "posix threads". 

I have no further idea...try to google the error: `/usr/bin/ld: cannot find -lpthreads`. That's what I would do.

closed 2023-05-10 by jbehley