jbehley/SuMa #44: error when ./visualizer

ContinueLopo, 2023-07-17:
Dear author：

When I finished compiling and was ready to run visualizer, the following error occurred

```
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
Segmentation fault (core dumped)
``` 

So I use the gdb command,gdb ./visualizer

```
GNU gdb (Ubuntu 10.2-0ubuntu1~20.04~1) 10.2
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./visualizer...
``` 

then (gdb) run

```
Starting program: /home/gao/robot/src/semantic_suma/bin/visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffb5fe9000 (LWP 38225)]
[New Thread 0x7fffb55c2000 (LWP 38226)]
OpenGL Context Version 4.6 core profile
GLEW initialized.
OpenGL context version: 4.6
OpenGL vendor string  : NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2

Thread 1 "visualizer" received signal SIGSEGV, Segmentation fault.
Frame::Frame (this=0x5555560aac50, w=<optimized out>, h=<optimized out>) at /home/gao/robot/src/semantic_suma/src/core/Frame.h:32
32	  residual_map(width, height, glow::TextureFormat::RGBA_FLOAT)
``` 

at last (gdb) bt

```
#0  Frame::Frame (this=0x5555560aac50, w=<optimized out>, h=<optimized out>) at /home/gao/robot/src/semantic_suma/src/core/Frame.h:32
#1  0x00005555555b9aa1 in __gnu_cxx::new_allocator<Frame>::construct<Frame, int, int> (this=<optimized out>, __p=0x5555560aac50) at /usr/include/c++/9/new:174
#2  std::allocator_traits<std::allocator<Frame> >::construct<Frame, int, int> (__a=..., __p=0x5555560aac50) at /usr/include/c++/9/bits/alloc_traits.h:483
#3  std::_Sp_counted_ptr_inplace<Frame, std::allocator<Frame>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<int, int> (__a=..., this=0x5555560aac30)
    at /usr/include/c++/9/bits/shared_ptr_base.h:548
#4  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<Frame, std::allocator<Frame>, int, int> (__a=..., __p=<optimized out>, this=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:679
#5  std::__shared_ptr<Frame, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<Frame>, int, int> (__tag=..., this=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:1344
#6  std::shared_ptr<Frame>::shared_ptr<std::allocator<Frame>, int, int> (__tag=..., this=<optimized out>) at /usr/include/c++/9/bits/shared_ptr.h:359
#7  std::allocate_shared<Frame, std::allocator<Frame>, int, int> (__a=...) at /usr/include/c++/9/bits/shared_ptr.h:702
#8  std::make_shared<Frame, int, int> () at /usr/include/c++/9/bits/shared_ptr.h:718
#9  ViewportWidget::ViewportWidget (this=0x555555f807d0, parent=<optimized out>, shareWidget=<optimized out>, f=...) at /home/gao/robot/src/semantic_suma/src/visualizer/ViewportWidget.cpp:37
#10 0x00005555555df2c2 in Ui_MainWindow::setupUi (this=this@entry=0x7fffffffc568, MainWindow=MainWindow@entry=0x7fffffffc520) at /home/gao/anaconda3/include/qt/QtCore/qflags.h:123
#11 0x00005555555cf39d in VisualizerWindow::VisualizerWindow (this=0x7fffffffc520, argc=<optimized out>, argv=<optimized out>)
    at /home/gao/robot/src/semantic_suma/src/visualizer/VisualizerWindow.cpp:35
#12 0x0000555555583d88 in main (argc=<optimized out>, argv=0x7fffffffce88) at /home/gao/robot/src/semantic_suma/src/visualizer/visualizer.cpp:19
``` 

Can you please help me with this?

jbehley, 2023-07-19:
It's hard to tell what is the reason of this error. Therefore, some additional questions:

1. Do you use the the off-the-shelf configuration? Maybe it's the setting of the "width" and "height" of `Frame`?
2. Is it maybe a problem of the available VRAM? Do you have some other processes consuming additional memory used on the GPU by other programs, and there is simply not enough space?
3. Usually, the driver of Nvidia GPUs is quite forgiving regarding the "core" vs. "compatible" profile. At least, we have less problems with Nvidia GPUs. You could check to replace the `glow::TextureFormat::RGBA_FLOAT` with `glow::TextureFormat::RGBA` and see if it still occurs. Then it might be a problem with the driver.

ContinueLopo, 2023-07-20:
> It's hard to tell what is the reason of this error. Therefore, some additional questions:
>
>     1. Do you use the the off-the-shelf configuration? Maybe it's the setting of the "width" and "height" [...]
Thanks for the reply, I compiled and installed gtsam-4.2a9 today by compiling and installing it and it solved the problem, so it could be the version of gtsam that is causing the problem!

closed 2023-07-20 by ContinueLopo