jbehley/SuMa #45: some problems in running and using SuMa

zcc00210, 2024-04-25:
Thank you for making your work open source. Amazing work!
After running the following command `cmake .. -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=410`, I couldn't find anything in the `bin` directory of the source folder of the project, as shown below:
[image]
[image]
So I wonder where to find `visualizer` in the `bin` directory. Could you tell me what's wrong in this process?
Thank you very much!

jbehley, 2024-04-25:
can you provide the output on the terminal; what happens after calling `cmake`?

zcc00210, 2024-04-25:
here is the first output when i `cmake`:
```
jy@jy-virtual-machine:~/SuMa/build$ cmake .. -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=410
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [SuMa] Fetching glow.
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.1.0") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem system 
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Enabling Nvidia OpenGL extensions.
-- Using OpenGL version 430.
-- [SuMa] Fetching GTSAM v4.2.0.
Scanning dependencies of target gtsam-populate
[ 11%] Creating directories for 'gtsam-populate'
[ 22%] Performing download step (git clone) for 'gtsam-populate'
Cloning into 'gtsam-src'...
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Cloning into 'gtsam-src'...
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Cloning into 'gtsam-src'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
-- Had to git clone more than once:
          3 times.
CMake Error at gtsam-subbuild/gtsam-populate-prefix/tmp/gtsam-populate-gitclone.cmake:31 (message):
  Failed to clone repository: 'https://github.com/borglab/gtsam.git'

make[2]: *** [CMakeFiles/gtsam-populate.dir/build.make:92: gtsam-populate-prefix/src/gtsam-populate-stamp/gtsam-populate-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/gtsam-populate.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

CMake Error at /usr/share/cmake-3.16/Modules/FetchContent.cmake:915 (message):
  Build step for gtsam failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FetchContent.cmake:1006 (__FetchContent_directPopulate)
  /usr/share/cmake-3.16/Modules/FetchContent.cmake:1047 (FetchContent_Populate)
  external/gtsam.cmake:18 (FetchContent_MakeAvailable)
  external/find_dependencies.cmake:4 (include)
  CMakeLists.txt:8 (include)
[... 85 lines cut ...]
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.2           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem system timer date_time 
-- [SuMa] Enabling Nvidia OpenGL extensions.
-- [SuMa] Using OpenGL version 430.
-- [SuMa] Fetching GTest.
-- Found Python: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Found Threads: TRUE  
-- Configuring done
-- Generating done

```