jbehley/SuMa #46: Visualizer not found

laurenzheidrich, 2024-05-06:
Hey there!

I have been trying to set up SuMa on my workstation with VNC. The building runs without any problems, see here:

Command: cmake .. -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=450 -DENABLE_NVIDIA_EXT=YES

Output:

cmake .. -DCMAKE_BUILD_TYPE=Release -DOPENGL_VERSION=330 -DENABLE_NVIDIA_EXT=YES
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [SuMa] Fetching glow.
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so   
-- Found GLEW: /usr/include (found version "2.1.0") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem system 
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Enabling Nvidia OpenGL extensions.
-- Using OpenGL version 430.
-- [SuMa] Fetching GTSAM v4.2.0.
-- GTSAM Version: 4.2a9
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.3.7
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find TBB (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES tbb tbbmalloc) (Required is at least version "4.4")
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.2.0
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Wrote /mnt/scratch/SuMa/build/_deps/gtsam-build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 9.4.0
--  CMake version                                    : 3.16.3
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/make
-- Build flags                                               
--  Build Tests                                      : Disabled
--  Build examples with 'make all'                   : Disabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Disabled
--  Build GTSAM unstable Python                      : Disabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            :  -O3 -DNDEBUG
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_11
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  Use System Eigen                                 : On (Using version: 3.3.7)
--  Use System Metis                                 : OFF
--  Using Boost version                              : 1.71.0
--  Use Intel TBB                                    : TBB not found
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : STL
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.2           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
CMake Warning at build/_deps/gtsam-src/cmake/HandleFinalChecks.cmake:3 (message):
  TBB 4.4 or newer was not found - this is ok, but note that GTSAM
  parallelization will be disabled.  Set GTSAM_WITH_TBB to 'Off' to avoid
  this warning.
Call Stack (most recent call first):
  build/_deps/gtsam-src/CMakeLists.txt:143 (include)

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem system timer date_time 
-- [SuMa] Enabling Nvidia OpenGL extensions.
-- [SuMa] Using OpenGL version 430.
-- [SuMa] Fetching GTest.
-- Found Python: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/scratch/SuMa/build

The problem is that the /mnt/scratch/SuMa/bin directory is empty, even though there is supposed to the Visualizer.

I would appreciate some help! Thanks!

laurenzheidrich, 2024-05-06:
This is the folder structure after the build:

[image]

jbehley, 2024-05-06:
I think you just did the "cmake .." and this will only produce the Makefiles for building everything.

you need to do inside the build directory now "make -j5" and this will then build the executable.

laurenzheidrich, 2024-05-06:
Well.. you are right and I should have paid more attention. Thanks, I will close this :)

closed 2024-05-06 by laurenzheidrich