jbehley/SuMa #47: about the input to Suma

Hello-Water, 2024-05-21:
Hello, thanks for your code releasing!
I have some questions about the input to Suma:
1) Does the `visualizer` default to using `config/default.xml` and how can i change config file to another one (for example `vlp16.xml`)?
2) If I just only feed Lidar data to Suma (there is no calib.txt and other files) , when I execute `Save Poses`, is the pose in output file poses.txt referenced to the first frame of the Lidar data?

jbehley, 2024-05-21:
thanks for your interest in our work and the feedback.
1.  Yes, I should have added it to the Readme; you call `./visualizer [<path/to/config>] [<path/to/data>]`, where the path to data is a directory containing the `.bin` files. config and path to data are optional directories.
2. Currently, the `savePoses` method doesn't consider the calibration (see [1], not sure when I did this...). So all poses are in the local reference frame of the first LiDAR.

It might be that stuff is quite fitted to KITTI (and their data format), but hopefully it will work our for you.

[1] https://github.com/jbehley/SuMa/blob/2d191f5a5c83e793ef2768611b2d5cad04943a9e/src/visualizer/VisualizerWindow.cpp#L840-L869

Hello-Water, 2024-05-22:
Okay, I got it. Thank you very much for your prompt reply!

Hello-Water, 2024-05-30:
@jbehley Hello, I have another question to ask for your advice:  whether the ego-motion compensation in the bin format of the KITTI dataset is compensated to the end of the lidar frame time?

jbehley, 2024-05-30:
Regarding what KITTI did for the motion compensation only the authors know. I guess they used the IMU estimates. But cannot tell for sure.